martinohanlon/BlueDot #16: Is position values going beyond -1 and 1 intentional?

bennuttall, 2017-03-30:
If you read the position's `x` and `y` coordinates, they can go outside the range of -1 to 1.

If this is intentional, what's the use case for that?

If it's not intentional, I suggest clamping the values in the property definition.

Trying to pass `x` and `y` into a gpiozero object's value is fine, until it hits a value outside of that range, and it falls over, so you have to clamp it yourself. I did this:

```python
def clamped(v):                                                                 
    return max(-1, min(1, v))
```

martinohanlon, 2017-03-31:
It's a symptom of how android manages objects on a screen, moved events and touch whereby once an object is pressed you can move outside that object (a little bit).

I experienced in during testing and i could never decide whether to clamp it at the client, library or perhaps for a level of future proofing both!

It's also worth noting that while the image is round  the button is actually square and you can go to the corners x=+1, y=+1.

I'll clamp it at one in the next change.

martinohanlon, 2017-03-31:
fixed in 0.0.4

closed 2017-03-31 by martinohanlon