martinohanlon/BlueDot #17: Read device orientation data from the app

bennuttall, 2017-03-31:
Quite a big request, but it would be awesome if you could read the device's orientation as well as the dot position.

martinohanlon, 2017-03-31:
Its a potential for the future.

This feels like a separate operation (3) returning orientation data.

https://developer.android.com/reference/android/hardware/SensorManager.html describes how to use the sensors to get orientation.

dglaude, 2017-06-02:
Rather than to report the device orientation, the app could adapt it self to the orientation.
If I put my phone sidewise (landscape) and I press up, it should be the same output as when in portrai and I press UP.

martinohanlon, 2017-06-02:
@dglaude this issue refers to the physical orientation of the phone (i.e. as reported by the gyroscope) not the orientation of the screen.

dglaude, 2017-06-02:
Then should I raise another issue?
The usecase is real, you may want to hold your phone one way or another.