martinohanlon/BlueDot #24: Setup travis testing

martinohanlon, 2017-04-03:


martinohanlon, 2017-04-03:
Need to do:

- remove Python - "3.2"
- sort out "ImportError: No module named dbus" error
- remove ^M from .travis.yml

martinohanlon, 2017-04-03:
Try adding 

sudo: required to .travis.yml

martinohanlon, 2017-04-06:
Sorted by putting dirty hack into utils.py to check for the travis environment variable and not import dbus if running by travis.

ukBaz, 2017-04-06:
system_site_packages only works with Python 2.7 and 3.2 on Travis.
If you select just those two versions then you can do apt-get python-dbus

I've also looked at using [python-dbusmock](https://github.com/martinpitt/python-dbusmock) for easing testing of DBUS on Travis

martinohanlon, 2017-04-06:
@ukBaz yes I found that out by trial and error and then looking at your travis setup!

I wasnt keen to only test with Python 3.2 as I read that bluetooth sockets where only supported from 3.3

martinohanlon, 2017-11-06:
Setting up travis was just a massive pain due to dbus. so just setup local testing using pytest.

closed 2017-11-06 by martinohanlon

ukBaz, 2017-11-07:
Hi @martinohanlon ,

It might be too little too late but FYI:
I've also had to abandon testing with dbus on Travis as it was too much pain.
My approach now is to unittest.mock.patch the DBus calls. e.g:
https://github.com/ukBaz/python-bluezero/blob/master/tests/test_adapter_example.py#L16