martinohanlon/BlueDot #39: Pairing a headless Pi

whaleygeek, 2017-04-08:
Could we have a feature that makes it possible to put a headless Pi into pairing mode (e.g. by pressing a GPIOzero button to make the device discoverable)

whaleygeek, 2017-04-08:
[image]

ukBaz, 2017-04-08:
GPIO button is good. If you want to do it headless over ssh then use bluetoothctl
[pi] discoverable on

martinohanlon, 2017-04-18:
I've been looking into this.

My suggestion is to add a method along the lines of `make_pairable` which will put the bluetooth device into discoverable and pairable mode, meaning that it will be visible and accept pairing requests. Some quick POC's suggest this is works.

Thoughts at the moment are:
 - create a method which makes the device pairable for a set period of time (e.g. 60 seconds)
 - create a read only property which returns whether the device is currently pairable.

What do you think?

martinohanlon, 2017-04-23:
in v0.2.0 http://bluedot.readthedocs.io/en/latest/recipes.html#pairing

closed 2017-04-23 by martinohanlon