martinohanlon/BlueDot #40: Keep getting import errors

erikvanelten, 2017-04-12:
Tried it on a headless Pi Zero W and on a Pi 3B 1.2, with both Python 2.7.9 and Python 3.4.2 but every time I get the following when I try to run the first bit of code:

Traceback (most recent call last):
  File "bluedot.py", line 1, in <module>
    from bluedot import BlueDot
  File "/home/pi/bluedot/bluedot.py", line 1, in <module>
    from bluedot import BlueDot
ImportError: cannot import name BlueDot

martinohanlon, 2017-04-12:
Have you run, and has it completed successfully, the install of python-dbus
and bluedot:

sudo apt-get install python3-dbus
sudo pip3 install bluedot

Are you running a recent version of Raspbian with pixel? Previous versions
of Raspbian didn't include the bluetooth libraries.

martinohanlon, 2017-04-12:
I see the problem....

You have called your program bluedot.py...

Now when you:

import bluedot

.. it's importing your bluedot.py program not the bluedot module as Python
looks in the local directory for a file called bluedot before looking in
the installed packages.

Rename your program and delete all the pycache & *.pyc directories and
files.

martinohanlon, 2017-04-12:
Put a note into the getting started instructions about not calling your program bluedot.

martinohanlon, 2017-04-12:
Updated getting started guide.

closed 2017-04-12 by martinohanlon

erikvanelten, 2017-04-12:
Martin,

Thanks for the quick reply!

Let's call this a beginner's error, but to be honest I think this is strange Python behaviour.

But this did the trick. Once, in fact. The BT connection was lost immediately, but that's probably due to the wobbly BT-connection with my Nexus 5.

Anyway: solved!!

Cheers,

Erik

crazyonej, 2018-06-29:
I am having the same issue as above,  "ImportError: No module named bluedot"

I followed the steps but get the same error on both a zero w and Pi3

pi@raspberrypiFULL:~/python $ python MyDot.py
Traceback (most recent call last):
  File "MyDot.py", line 1, in <module>
    from bluedot import BlueDot
ImportError: No module named bluedot
pi@raspberrypiFULL:~/python $ sudo apt-get install python3-dbus
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-dbus is already the newest version (1.2.4-1).
The following packages were automatically installed and are no longer required:
  lxkeymap python-cairo python-gobject python-gobject-2 python-gtk2 python-xklavier
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
pi@raspberrypiFULL:~/python $ sudo pip3 install bluedot
Requirement already satisfied: bluedot in /usr/local/lib/python3.5/dist-packages
pi@raspberrypiFULL:~/python $
pi@raspberrypiFULL:~/python $ python MyDot.py
Traceback (most recent call last):
  File "MyDot.py", line 1, in <module>
    from bluedot import BlueDot
ImportError: No module named bluedot