martinohanlon/BlueDot #44: Mock App disconnects

erikvanelten, 2017-04-13:
After launching the first sample script a connection (?) is made (blue dot appears) and immediately disconnects.
====================================
pi@raspberrypi_3:~/bluedot $ python m.py
Server started 00:00:00:00:00:00
Waiting for connection
Client connected 11:11:11:11:11:11
Client disconnected
====================================

martinohanlon, 2017-04-13:
Can you provide your script please.

erikvanelten, 2017-04-13:
It's the first saample script from the recipes ( http://bluedot.readthedocs.io/en/latest/recipes.html#mock-app )

from bluedot import MockBlueDot

def say_hello():
    print("Hello World")

bd = MockBlueDot()
bd.when_pressed = say_hello

bd.mock_blue_dot_pressed()

martinohanlon, 2017-04-13:
I see the problem.

I tested running from IDLE which keeps the application alive. If you run from the command prompt as soon as the mock app has finished launching the program ends which simulates the client disconnecting.

If you want to run it via the command prompt you will need to use pause() to keep your program alive.

```
from bluedot import MockBlueDot
from signal import pause

def say_hello():
    print("Hello World")

bd = MockBlueDot()
bd.when_pressed = say_hello

bd.launch_mock_app()

pause()
```

Ill update the docs.

Also as an aside, I would recommend running your programs using python 3, i.e.

    python3 m.py

rather than:

    python m.py

which runs as python 2.

erikvanelten, 2017-04-13:
Fantastic! Works like a charm now! Great stuff, this. I really like it a lot, so much potential.

and btw: sorry for the strange font size above, I think I accidentally touched a button....

erikvanelten, 2017-04-14:
Off-topic, but I think you'd like to know this. I installed everything (IDLE on Python3, Pygame, BlueDot etc.) on my laptop (Intel i5, 6 GB running Linux Mint) and the Mock App runs well, but takes almost 98% CPU... My laptop overheated and shut down.

?

martinohanlon, 2017-04-14:
Interesting. pygame must be doing something different with events. The Pi seems to wait until events occur rather than looping.

I have updated the mock blue app  (v 0.1.2) to implement a clock, hopefully that will make a difference, its a better implementation.

I have also updated the docs re the original issue.

closed 2017-04-14 by martinohanlon