martinohanlon/BlueDot #54: Better report off duplicate use by catching the exception.

dglaude, 2017-06-02:
One copy of my program was running in the background (maybe in a screen) and I did notice.
While launching a second instance I got the error below.
Maybe the error message could be more explicit to help troubleshoot it by the user.

Traceback (most recent call last):
  File "r.py", line 21, in <module>
    bd = BlueDot()
  File "/usr/local/lib/python3.4/dist-packages/bluedot/dot.py", line 401, in __init__
    self.start()
  File "/usr/local/lib/python3.4/dist-packages/bluedot/dot.py", line 600, in start
    self._server.start()
  File "/usr/local/lib/python3.4/dist-packages/bluedot/btcomm.py", line 340, in start
    self._server_sock.bind((self.server_address, self.port))
OSError: [Errno 98] Address already in use

martinohanlon, 2017-06-04:
note - would also need to check for python 2 error  

```
Traceback (most recent call last):
  File "examples/dot_debug.py", line 35, in <module>
    dot.start()
  File "/usr/local/lib/python2.7/dist-packages/bluedot-0.4.0-py2.7.egg/bluedot/dot.py", line 600, in start
    self._server.start()
  File "/usr/local/lib/python2.7/dist-packages/bluedot-0.4.0-py2.7.egg/bluedot/btcomm.py", line 340, in start
    self._server_sock.bind((self.server_address, self.port))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use

```

martinohanlon, 2017-06-04:
Added message 

```
Bluetooth address B8:27:EB:FE:0C:0E is already in use - is the server already running?
```

martinohanlon, 2017-06-04:
in dev v1.0.0

closed 2017-06-04 by martinohanlon