martinohanlon/BlueDot #56: Enhancement request: Rotary dial

ukBaz, 2017-06-02:
I've been having a play and was wondering if you could add something to make BlueDot behave like a rotary dial.

The use model that I have in mind is something like:
- swipe down turns on device (e.g. lamp)
- swipe up turns off device
- swipe left or right changes through menu. (E.g Adjust R, G or B)
  - When in menu mode you have 10 seconds to start drawing a circle (e.g. rotary dial mode).
  - If you draw anti clockwise the value goes down
  - if you draw clockwise the value goes up.
  - When specifying rotary dial you could give it a min and max value.

Interested to hear your feedback.

Thanks

martinohanlon, 2017-06-03:
This sounds like issue #11 ?

ukBaz, 2017-06-04:
I have been looking at this a little more this morning. Not sure where you are with implementing the click wheel functionality so I thought I would share where I am with it.

What I have not been able to implement is switching between modes. i.e. when is it a swipe/click versus drawing a circle.
What I have been able to do is implement a function that returns a negative number for anticlockwise and positive for clockwise. The function assumes you haven't moved more than 180 degrees between readings. In my tests I couldn't move that fast so it might be an OK assumption.

My function is:

```python
def calc_rotation(prev, current):
    delta = current - prev
    if -180 < delta <= 180:
        pass
    elif delta > 180:
        delta = delta - 360
    elif delta <= -180:
        delta = delta + 360
    return delta
```

An example use is:
```python
from bluedot import BlueDot
from signal import pause

bd = BlueDot()
prev_angle = 0

def calc_rotation(prev, current):
    delta = current - prev
    if -180 < delta <= 180:
        pass
    elif delta > 180:
        delta = delta - 360
    elif delta <= -180:
        delta = delta + 360
    return delta

def move(pos):
    global prev_angle
    current_angle = pos.angle
    turn = calc_rotation(prev_angle, current_angle)
    print('Moved ', turn)
    prev_angle = current_angle

bd.when_moved = move

pause()
```

I did some tests as I was developing it:
```python
import unittest
from click_wheel.click_wheel import calc_rotation

class GetRotation(unittest.TestCase):
    def test_anti_clk(self):
        self.assertEqual(calc_rotation(120, 100),
                         -20)

    def test_clkwise(self):
        self.assertEqual(calc_rotation(100, 120),
                         20)

    def test_bndry_anti_clk(self):
        self.assertEqual(calc_rotation(10, -10),
                         -20)

    def test_bndry_clkwise(self):
        self.assertEqual(calc_rotation(-10, 10),
                         20)

    def test_south_bndry_anti_clk(self):
        self.assertEqual(calc_rotation(-170, 170),
                         -20)

    def test_south_bndry_clkwse(self):
        self.assertEqual(calc_rotation(170, -170),
                         20)

if __name__ == '__main__':
    unittest.main()

```

I hope this helps move this issue forward in some way.

Thanks

martinohanlon, 2017-12-09:
I have created a POC - https://github.com/martinohanlon/BlueDot/blob/dev/examples/click_wheel.py

It seems to work quite well, it needs some refinement, but its 90% there.

It works by splitting the dot into segments, when you pass from 1 segment to the next it is considered to have been rotated.

Im thinking a `when_rotated` property which will call a function and pass a BlueDotRotation object (similar to when_pressed, when_swiped).

BlueDotRotation will have properties such as:

- `.value` - -1 for anti-clockwise, 1 for clockwise
- `anti_clockwise` - True if rotated anti clockwise
- `clockwise` - True if rotated clockwise
- `previous_segment` - no between 1 & no_of_segments which returns where the position has moved from
- `current_segment` - no between 1 & no_of_segments which returns where the position has moved too
- anything else I think of at the time.

martinohanlon, 2017-12-10:
Included in v1.2.0

closed 2017-12-10 by martinohanlon