martinohanlon/BlueDot #63: Multiple user?

dglaude, 2017-06-17:
Hi, this is a feature request... or a recepie request if it is already possible.

We (dad and son) would like to be able to connect more than one BlueDot app to a python program.

I don't know if it is technically possible.
I don't know how the API can be modified to support that scenario without breaking compatibility.

But I can see some use for it:
* Control more motor on a robot (like the pilote controling the move and co-pilote controling the camera heading)
* Shared control where the co-pilote can take over the control

As for implementation, here is some idea:
* pos.player will contain 1 for first or single controler
* pos.player will contain 2 for second controler

If possible, the app will display if you are player 1 or player 2.

The reason I say player is that of course BlueDot could be use to control a video game.

martinohanlon, 2017-06-19:
I suppose this is possible...  Although I have never tried.

You will have to have 2 bluetooth devices though i.e. 1 on board Bluetooth and a usb dongle, or 2 usb dongles because once connected Bluetooth is only one to one.

You would have to pair phone 1 to bluetooth device 1 and phone 2 to bluetooth device 2 - I suspect you would have to use the bluetoothctl command line utility because I doubt the GUI is multi-device aware (although as I said I havent tried, so it might be).

The code would be pretty simple though.

```
bd0 = BlueDot(device = "hci0")
bd1 = BlueDot(device = "hci1")
```

dglaude, 2017-06-19:
I was more hoping a two clients on the same BT interface, the same way I
can get a mouse, a keyboard and a speaker.

martinohanlon, 2017-06-19:
Unfortunately when you create an RFCOMM connection

martinohanlon, 2017-06-19:
Unfortunately when you create an RFCOMM connection it's a 1 to 1.

ukBaz, 2017-06-20:
While RFCOMM is a 1 to 1 connection. You can have multiple RFCOMM connections, right?

martinohanlon, 2017-06-20:
I (think - I havent tested), you can have multiple RFCOMM connections but they have to be on separate ports.  The Blue Dot app (both on them) use a set port (1).

Assuming you can have multiple connections over RFCOMM, to achieve this it would require a 'settings' page in the app where you could change the default port.  

If this change was made to the app, you would interact with it like this:

```
bd1 = BlueDot(port = 1)
bd2 = BlueDot(port = 2)
```

I am in 2 minds as to whether this is the right thing to do.  The first objective for Blue Dot is that its simple, it delivers 80% of the use cases for 20% of the complexity.  Would creating a settings page result in greater errors (users will undoubtedly change it) for little benefit? Is there a better way to support multiple devices?  The API doesnt really work for multiple devices, it is meant to be a 1 to 1 relationship.

TheDiveO, 2017-06-29:
Would trying the port number within a certain low range at least allow for a few blue buttons to be connected simultaneously?

martinohanlon, 2017-09-09:
So I have been looking into this. What I was hoping to do was get the Blue Dot app to automatically find the channel (port) (rather than having a 'pick a port' function).

It seems feasible to allow more than 1 connection through a single device using RFCOMM channels.

There are a couple of problems. Android uses SDP to determine which channel to use when connecting to a Bluetooth service via its UUID (you cant explicitly set a channel when making a connection). 

The bluedot library can register a different port to the UUID in SDP when it creates a Bluetooth server. All good, you can create a BlueDot listening on port 2.

The problem comes in that bluez will only allow you to register 1 profile per UUID, so I cant register multiple channels against a single UUID.

More investigation is needed

martinohanlon, 2017-09-09:
I have pushed some changes into dev to support multi user but you need more than 1 bluetooth device e.g. 

    bd0 = BlueDot(device = "hci0")
    bd1 = BlueDot(device = "hci1")

This works :)

martinohanlon, 2017-11-01:
I am going to close this issue. Conflicting constraints in bluez and android make multi-user support through 1 bluetooth device not possible while maintaining Blue Dot's simple interface.

closed 2017-11-01 by martinohanlon

LtPinback, 2018-11-04:
I was able to do this:

```
bd1 = BlueDot(port = 1)
bd2 = BlueDot(port = 2)
```

 via changing the port number in the android client with this modified [code](https://pastebin.com/T4ciHssE) and rebuilding the app. Sadly my Java and Android knowledge is zero so I don't know how to implement a dialog on the Devices.java page to select a port number before connection takes place. I share it here in the hope that Martin or someone else find it useful and somehow implements a way to do it in the app. Right now I have to change the number manually in the Android Studio and build a different app for each port.

martinohanlon, 2018-11-04:
Thanks for the additional information. 

I knew it was possible having previously created a prototype.

I didnt want to include it in BlueDot as the intention for the project is to be as simple as possible. 

If it were possible for Bluez to advertise multiple ports and Android automatically select the port (as it does for a single device), I would be open to implementing this change.

martinohanlon, 2020-01-04:
@LtPinback @dglaude 

I thought you would like to know I have recently added this feature to the android Blue Dot app.

https://bluedot.readthedocs.io/en/latest/recipes.html#multiple-blue-dots

The rule of 3 is a good approach, 3 people ended up asking for this so I thought it would be good time to include.

The approach I took was to include a settings page and make it an explicit action to turn off "auto port discovery" under an "advanced configuration" section. The app will also including updates to the on-screen messages which make it clear the user is using a specific port.

LtPinback, 2020-01-05:
Cool. I have learned the bare minimum since last year and was able to build my own client app that suits my user application (controlling a home alarm) but I am sure this could be very useful for those with zero knowledge in programming. Multiple clients after all is a pretty common thing for automation and the like.