martinohanlon/BlueDot #67: Exception thrown when callback function uses Python3 return value declaration

TheDiveO, 2017-07-05:
When I define a callback/event function as:
```python3
bd_pressed() -> None:
    ...
```
then bluedot/dot.py fails in `_process_callback` (or more precisely, in `getargspec`) with the error message "ValueError: Function has keyword-only arguments or annotations, use getfullargspec() API which can support them".

martinohanlon, 2017-07-05:
Can you give me a little more information please.  

What are you trying to achieve?

Can you provide a full code listing of how to recreate the error as well please.

martinohanlon, 2017-07-05:
Note - Migrate to use getfullargspec()

TheDiveO, 2017-07-07:
```python
import bluedot
bd = bluedot.BlueDot()

def bd_pressed() -> None:
    pass

bd.when_pressed = bd_pressed
```

martinohanlon, 2017-07-23:
fixed in v1.0.2

closed 2017-07-23 by martinohanlon