martinohanlon/BlueDot #96: Client to Server Communication

Stamenov96, 2018-01-19:
Hi,
I'm using the BlueDot app for controlling my RC car. I have BlueDot python server and the BlueDot android app. I have changed a little the android app, i have just added a new activity with two input fields. Now, i'm trying to send the information from the input fields to the python server.Could you help me to do this?

martinohanlon, 2018-01-19:
I can give you some pointers but I am not going to be able to make this change.  The whole point of blue dot is that it is a simple interface with 1 function, it is a solution to get started.

If you look at the button activity in the BlueDot:

[https://github.com/martinohanlon/BlueDot/blob/master/clients/android/app/src/main/java/com/stuffaboutcode/bluedot/Button.java](https://github.com/martinohanlon/BlueDot/blob/master/clients/android/app/src/main/java/com/stuffaboutcode/bluedot/Button.java)

1. It creates a [connect to bluetooth](https://github.com/martinohanlon/BlueDot/blob/master/clients/android/app/src/main/java/com/stuffaboutcode/bluedot/Button.java#L50)

2. When touch events are detected it calls [pressed, released, moved functions](https://github.com/martinohanlon/BlueDot/blob/master/clients/android/app/src/main/java/com/stuffaboutcode/bluedot/Button.java#L71)

3. These functions construct a message which conforms to the format [operator], [x], [y]\n

4. They then use the [send function to pass the message to the server](https://github.com/martinohanlon/BlueDot/blob/master/clients/android/app/src/main/java/com/stuffaboutcode/bluedot/Button.java#L114)

These messages end up being managed by dot.py in bluedot python library
[https://github.com/martinohanlon/BlueDot/blob/master/bluedot/dot.py](https://github.com/martinohanlon/BlueDot/blob/master/bluedot/dot.py):

1. When the server receives from data it passes it to [BlueDot._data_received](https://github.com/martinohanlon/BlueDot/blob/master/bluedot/dot.py#L898)

2. _data_received makes sure only complete messages (those which end in \n) are processed and passes them to [_process_commands](https://github.com/martinohanlon/BlueDot/blob/master/bluedot/dot.py#L910)

3. _process_commands splits the command into its component parts (operator, x, y) and calls the relevant function to state whether it was pressed, moved, released.

You could feasibly change the app to send a new message with a different operator and modify _process_commands to process that new operator in whatever way you wanted.

I hope that helps.

martinohanlon, 2018-01-19:
What additional data do you want to send?

closed 2018-01-19 by martinohanlon

Stamenov96, 2018-01-19:
A string type data, for example Name of the home wi-fi Network and it's password.

Stamenov96, 2018-01-24:
It's me again :laughing: 
I've changed some things in the android app so I send a command with code 3.
I've added the case 3 in dot.py and i think it should work, but how can I recompile the project and installing it. I've tried with 
setup.py clean
setup.py build
setup.py install
But when i call my function it says "SyntaxError: ivalid syntax" is there something that i have missed with the build and install process, or the error is somewhere in my function/code/dot.py ?
Is there any possible way to debug just dot.py ?

martinohanlon, 2018-01-27:
If you are receiving syntax error it will be because you have an error in your code.

You can use `setup.py develop` to install a development code of a python library.