martinohanlon/BlueDot #99: how to change UUID?

monkey-jsun, 2018-02-12:
First of all, great code and great feature!

I have been looking for simple rfcomm based bt communication with Android phones and btcomm is a really nice and full-featured library for that.  I pretty much got everything integrated in my project.

Now I'd like to change UUID so that it does not collide with bluedot.  However, I found out simply changing it to a new one (and changing on Android side) does not work.  What is the gotcha here?

Thanks!

martinohanlon, 2018-02-12:
Where did you change it? [utils.register_spp](https://github.com/martinohanlon/BlueDot/blob/master/bluedot/utils.py#L101) and the [button activity in the android app](https://github.com/martinohanlon/BlueDot/blob/master/bluedot/utils.py#L101).

If so, that is all it should need.

martinohanlon, 2018-02-12:
Can you elaborate on 'does not work'?

martinohanlon, 2018-02-15:
Im closing this as its not an issue with Blue Dot. Let me know if you need anything though.

closed 2018-02-15 by martinohanlon