martinohanlon/BlueDot #104: Blue Dot drops connection immediatly

jenglows, 2018-03-25:
I can pair my Samsung 8 with my raspberry pi3 and it looks all good.  But when I open the Blue Dot app and choose the Pi3 from the list of available devices it immediately drops the pairing and gives me a "failed to connect" error.
I've tried most of the easy things I can think of to try to get it to work, but I must be missing something.  When I try to connect through the command line on the Pi3,  I can watch the changes between the connected yes and no when the app drops the pairing.

ukBaz, 2018-03-25:
It sounds to me like you don't have your BlueDot code running on the Raspberry Pi before you try to connect.

Do you get the code running and then try to connect as documented here:
http://bluedot.readthedocs.io/en/latest/gettingstarted.html#code

If you don't do that the app running on the Phone cannot find the right Bluetooth profile on the Raspberry Pi so disconnects.

martinohanlon, 2018-03-26:
What @ukBaz said..

If your python program is running and you still can't connect, does Python return an error? Do you get the client connected message? Can you copy your python code into the issue.

martinohanlon, 2018-04-09:
@jenglows how did you get on?

closed 2018-04-30 by martinohanlon

bshreyas13, 2020-07-15:
Hi ,
was this issue resolved? if so, how?
I am having the exact same issue.

Thanks

ukBaz, 2020-07-15:
Hi @bshreyas13 ,

People normally see this issue if they haven't started their script on the Raspberry Pi bafore trying to connect from their phone.

It works like this:

 - pair your phone and raspberry pi
 - write your Python program
  - RUN your Python program
  - when it is running
  - open the Blue Dot app on your phone
   - select your Raspberry Pi from the list of devices on the Blue Dot app on your phone

This is documented in the getting started guide. https://bluedot.readthedocs.io/en/latest/gettingstarted.html

bshreyas13, 2020-07-15:
Hey @ukBaz ,

I ran the example code from the terminal, it shows my devices MAC address then says waiting for connection . But still doesnt connect.

Thanks
Shreyas

bshreyas13, 2020-07-16:
[image]

Here you can see I have the script running and waiting. but the app doesn't connect on the phone.
I have pared it as well without problems.

thanks

bshreyas13, 2020-07-16:
I updated Debian and python-bluez which seems to have solved the issue.

Thanks for the inputs guys!