martinohanlon/BlueDot #107: Raspberry Pi to Raspberry Pi not connecting

Zxtreme03, 2018-04-30:
I am getting Error:  {Erno 111}   Connection Refused

martinohanlon, 2018-04-30:
Can you give me a little more information?

What are you doing? Are you using 2 Pi's? 1 Pi + Android phone?
What does your code look like?

Generally `connection refused` means that you are trying to connect and there is nothing to connect to.
 - is your blue dot program running on your Pi?
 - have you paired your Pi?
 - are you connecting to your Pi and not something else?

Zxtreme03, 2018-05-01:
I am connecting 2 raspberry Pis together. And the program is running.

martinohanlon, 2018-05-01:
Ok, so you are running `bluedotapp` on 1 raspberry pi, and your "server" python program on the other?

Can you share your code? Screenshots of both Pi's would be useful.

martinohanlon, 2018-05-27:
No response, closing. Feel free to reopen if you still have issues.

closed 2018-05-27 by martinohanlon