martinohanlon/BlueDot #110: Run on Boot and Reload if Closed

jerryfudd, 2018-06-25:
Hi,

I've got the BlueDot running as a joystick and working great but I'm having some issues making it run on boot up and re-open if quit so that I don't have to have a computer involved in the setup.

Tried running the cron below which doesn't do the job - but runs fine when pasted into terminal (minus the asterix's of course).

`* * * * * if ! ps -ef | grep -v "color" | grep "sudo python /usr/local/bin/control_bluetooth.py"; then sudo python /usr/local/bin/control_bluetooth.py; fi`

any ideas on how to achieve this? ....not necessarily in cron

ukBaz, 2018-06-25:
If this works on the command line but not in crontab then the things to investigate are things like what shell is used in crontab versus your shell. And the environment variables in crontab versus your shell.

However, rather than investigate those, the approach I would suggest is to simplify the crontab entry and move the complexity into a new shell script. So for example, run the new script every 30 seconds from crontab:
```
*/30 * * * * /home/pi/bluedot_launch.sh
```

Then the bluedot_launch.sh script would be something like the following
(You can choose the type of shell with the [shebang line](https://en.wikipedia.org/wiki/Shebang_(Unix)))
```
#!/bin/sh
if ! ps -ef | grep -v "color" | grep "sudo python /usr/local/bin/control_bluetooth.py"; then 
    sudo python /usr/local/bin/control_bluetooth.py; 
fi
```
Make sure the script is executable (chmod +x /home/pi/bluedot_launch.sh) then you can test from the command line with `/home/pi/bluedot_launch.sh`

If that works then I would be more confident of it working in crontab.

martinohanlon, 2018-06-26:
I tend to put a script in `/etc/rc.local`.

add:

```bash
python /usr/local/bin/control_bluetooth.py &
```

NOTE - remember the final & otherwise your Pi will stop booting waiting for your program to finish - bas news!

before the `exit 0`

Im closing because its not an issue with BlueDot but feel free to ask any questions.

closed 2018-06-26 by martinohanlon

jerryfudd, 2018-06-26:
thanks, keep forgetting about rc.local

I chucked the 'if' in a script and then stuck that in the cron as ukBaz suggested - it starting the script on boot now and re-opening if it doesn't detect the process but will have to wait till I get home to check if I can connect to the bluetooth.