martinohanlon/BlueDot #112: Blue Dot Drops connection immeadity

HughEvans01, 2018-07-14:
Whenever I run a script it instantly closes like so:
`pi@raspberrypi:~/LearningBlueDot$ python3 example2.py`
` Server started B8:27:EB:05:3A:5A`
`Waiting for connection`
`pi@raspberrypi:~/LearningBlueDot$`
Attempting to connect via the app always returns "Failed to connect". I've restarted my Pi 3 several times and have disconnected and reconnected my phone using Bluetoothctl several times to no avail - any ideas?

HughEvans01, 2018-07-14:
Think I might have resolved this issue, or at least in my case. On the Blue Dot API documentation the "when_pressed" example is as follows:
`from bluedot import BlueDot`
`def dot_was_pressed():`
`    print("The Blue Dot was pressed")`
`bd = BlueDot()`
`bd.when_pressed = dot_was_pressed`
Which had the issue I described above, however; "bd.when_pressed" alone is not a valid event listener (for me at least) - it seems to have various issues. This code, however; works although sometimes it will take several attempts to connect properly.
`from bluedot import BlueDot`
`def dot_was_pressed():`
`    print("The Blue Dot was pressed")`
`bd = BlueDot()`
`while True:`
`    bd.when_pressed = dot_was_pressed`
Is this an issue others have had or is it just my setup?

ukBaz, 2018-07-14:
Hi @HughEvans01,

As a side note you might want to take a look at the following GitHub guide so that your code snippets are a little more readable for people: 
https://help.github.com/articles/creating-and-highlighting-code-blocks/

The following example from the documentation should allow you to connect your Android phone. It will display the message on the RPi before exiting.
```python
from bluedot import BlueDot
bd = BlueDot()
bd.wait_for_press()
print("You pressed the blue dot!")
```
This is a great place to start as this will prove you have everything setup and working.

If you want to use `when_pressed` then you need to do a little bit more. From the example recipes 
```python
from bluedot import BlueDot
from signal import pause

def say_hello():
    print("Hello World")

bd = BlueDot()
bd.when_pressed = say_hello

pause()
```

Try those and let us know how you are getting on

HughEvans01, 2018-07-14:
Did the above already, you guys might want to add "pause" to your API documentation examples as without it some of the examples don't work. Thanks for the advice

ukBaz, 2018-07-14:
So is this working for you now?