martinohanlon/BlueDot #115: Avoid new connections at time

manuelpt4940, 2018-11-01:
Good day,

When I have connected with server Bluetooth, the server mantain open to receive new connections, even if I'm connected (obviously if other dispositives have the credentials - MAC - to connect with server). How I can close the server to avoid receive connections until I disconnect myself with server? In other words, Can I enable only 1 connection at time?

martinohanlon, 2018-11-01:
I'm not sure I understand your question / issue.

If you are using Blue Dot you can only establish 1 connection at a time.

Do you mean, how can I ensure that only 1 specific device can connect?

martinohanlon, 2018-11-04:
@manuelpt4940 can you provide any additional information?

manuelpt4940, 2018-11-05:
Sorry for the late. I used the server Bluetooth in my Pi3, so I connected my cellphone and all was ok, however, I test with other cellphone and I could connect too, but if i send data with the second cellphone, the server doen't show data in the screen, only show data that i sent trhoughout my first cellphone, but when i disconnected first device, all data sent with cellphone 2 showed in the server, like if the buffer the server store all data and after the first device was disconnected, so it would show it. This test I made it some months ago, I don't know if this inconvenient is still occurrence, but I will test it again.

manuelpt4940, 2018-11-07:
Effectively, I repeat test, and the server allow connect 2 cellphone at time, and the second can send data but the first cellphone has priority, however, when i disconnect the first cellphone, the second receive the prority and the server in raspberry show all data sent througout cellphone 2. Can I close server when it receive a conection? and open it again when this connection is finished?

martinohanlon, 2018-11-07:
Thanks for the additional information. In short this shouldn't happen and
didn't use to happen.

I'm wondering whether something has changed in the most recent version of
bluez.

I'll do some tests.

martinohanlon, 2018-11-08:
I can replicate the issue, how very strange. Off the top of my head I cant think of a way to restrict it using the current API as the 2nd connection doesnt trigger a "client connected" event.

I will look at a fix / api change. 

Interestingly the if rfcomm is no longer restricted to a single connection on a single port it does open up some interesting opportunities for using multiple blue dots. In fact there is a closed issue discussing this.

martinohanlon, 2018-11-08:
Ok, so I have done some testing... 

This is what is happening:

- BlueDot create a socket server which waits for bluetooth connections which involves:
  - creating a socket
  - binding a socket
  - listening for connection
- BlueDot then waits for a connection
- Client 1 connects, the connection is accepted
- BlueDot stops accepting new connections and reads from client 1
- if Client 2 connects, the socket connect is established and the Android app shows the blue circle, but the BlueDot python library doesn't accept the socket because it is busy reading data from Client 1
- When Client 1 disconnects BlueDot goes make to accepting new connections, accepts the new connection from Client 2 and starts reading data from that.

I made some changes to test what was happening and this is what I found.
 - Firstly the BlueDot python library doesn't know a client has connected until it tries to accept the connection
 - As soon as BlueDot accepts a connection from Client 2, the connection to Client 1 is closed (this is due to Bluetooth only supporting 1 connection on 1 port)
 - The Android app cant tell if its connection has been accepted, it only knows it has connected.

So, it would be possible to disconnect Client 1 when Client 2 connects, but this doesnt feel like the right solution, in my mind the Client which connected first should have the connection until it drops.

A "typical" way of achieving this would be to have a "handshake" i.e. when a client connects it sends a message saying "hi, im here", the server would then respond with "hi, you are connected", if the client didnt receive anything it would disconnect. Unfortunately comms between the app and the server are current 1 way only and it is a significant amount of work to make it 2 way!

Tto answer your question - no you cant restrict 1 connection at a time with BlueDot as it stands today.

I will leave this issue open as 2 way comms is something I want to implement into BlueDot at some point.

ukBaz, 2018-11-08:
Just to put in another idea on this... 
As part of the D-Bus API there is a `NewConnection` function that gets called when a new service level connection has been made. More details at:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/profile-api.txt#n119

I wondered whether that would allow you any more control?

I have an example of using the NewConnection function at:
https://gist.github.com/ukBaz/217875c83c2535d22a16ba38fc8f2a91

martinohanlon, 2018-11-08:
@ukBaz - interesting, thanks

manuelpt4940, 2018-11-08:
@martinohanlon handshake is a great option. I will implement it in my project to avoid others connections.