martinohanlon/BlueDot #124: Bluedot Android app with more buttons etc..

lectronix, 2019-02-06:
Hi,

It's great, thanks but how do I make an Android app with more than one Bluedot button or other control? A simple example would do.

Also is there an iOS version thought about?

Thanks

Kim@Lectronix

ukBaz, 2019-02-08:
FYI: I have been working on a Python BlueZ BLE UART implementation.
https://github.com/ukBaz/python-bluezero/blob/ble_uart/examples/ble_uart.py
I think your concern is that the user will need to have BlueZ in experimental mode and have modified the DBus configuration file to open the permissions up. e.g. ‘ukBaz.bluezero’ in my example

When Debian Buster is released it will contain a newer version (5.50) of BlueZ that will no longer require BlueZ to be in experimental mode for this type of functionality to work.

martinohanlon, 2019-02-08:
Yes @ukBaz the challenge is to deliver a simple installation. Good news about Buster.