martinohanlon/BlueDot #126: RC Car

mishu28nmv, 2019-03-07:
Hy to all.
I'm new to coding and RC World.
I have a RC Truck that I want to upgrade with the H Bridge and the RPi 3B+.
Tryed with 2 motors and the Robot example, but I have only 2 'controls:
pressing **top or left**: the 2 motors go clock way
pressing **bottom or right**: the 2 motors go counterclock way.
I want to:
press **top**:motor 1 goes clock way
press **bottom**:motor 1 goes counterclock way
press **left**:motor 2 goes clock way
press **right**:motor 2 goes counterclock way. Like in a normal RC Car not a Robot.

**Thank you for all your effort!**

mishu28nmv, 2019-03-15:
Never mind, did it on my own. Here is the code for a RC Car:
from bluedot import BlueDot
from gpiozero import Motor
from signal import pause

# moving pins 23,24; steering pins 17,22

bd = BlueDot()
bd.border = True
motor1 = Motor(forward=23, backward=24)
servo = Motor(forward=17, backward=22)

def move(pos):
    if pos.top:
        motor1.forward()
    elif pos.bottom:
        motor1.backward()
    elif pos.left:
        servo.forward()
    elif pos.right:
        servo.backward()        

def stop():
    motor1.stop()
    servo.stop()

bd.when_pressed = move
bd.when_moved = move
bd.when_released = stop

pause()

martinohanlon, 2019-03-15:
Well done.

closed 2019-03-15 by martinohanlon

mishu28nmv, 2019-03-15:
> Well done.

Thank you. Can you tell me what is the command to split the dot in 4 zones?
Something like this? [https://9to5google.com/wp-content/uploads/sites/4/2018/08/mdx_ic_dpad.jpg?quality=82&strip=all&w=668](url)

mishu28nmv, 2019-03-15:
Saw that in your examples one time, but I can't find it anymore :(

martinohanlon, 2019-03-15:
I'm not sure what you mean, im the link you include is broken. I get URL is unresolved.

mishu28nmv, 2019-03-15:
Sorry for the link. Was a picture with a circle with 4 visible zones.
I have now the hole circle with border and I want to split it in 4 zones with border for each zone. So the up, down, left, right zones are visible thanks to the border.

martinohanlon, 2019-03-15:
There is no way of changing the app so it shows the top, bottom, left, right sections.

mishu28nmv, 2019-03-15:
Ok. Can we have multi touch? For example when I press up and right at the same time, the RC car goes forward and turns right?

martinohanlon, 2019-03-17:
The app supports multitouch but it only returns single positions for the dot.

I would suggest if you want to do something like "forward and right" you use the [angle](https://bluedot.readthedocs.io/en/latest/dotapi.html#bluedot.BlueDotPosition.angle) of where the dot was pressed. That way you cant tell if the dot is pressed in the top right, rather than just at the top. 

e.g.

```python
from bluedot import BlueDot
from gpiozero import Motor
from signal import pause

bd = BlueDot()
bd.border = True
motor1 = Motor(forward=23, backward=24)
servo = Motor(forward=17, backward=22)

def move(pos):
    if pos.angle > 22.5 and pos.angle < 67.5:
        motor1.forward()
        servo.backward()

def stop():
    motor1.stop()
    servo.stop()

bd.when_pressed = move
bd.when_moved = move
bd.when_released = stop

pause()
```

mishu28nmv, 2019-03-17:
Ok super. I will write all the comands for the other turns and I will upload it here. After that can you upload it here and to your website if you want. I think it will be a grea succes!
Thank you for all your help!

mishu28nmv, 2019-03-22:
Code for RC Car
```python
from bluedot import BlueDot
from gpiozero import Motor
from signal import pause

moving pins 23,24; steering pins 17,22

bd = BlueDot()
bd.border = True
motoras = Motor(forward=23, backward=24)
servo = Motor(forward=17, backward=22)

def move(pos):
    if pos.angle > -22.5 and pos.angle < 22.5:
        motoras.forward()
    elif pos.angle > 22.5 and pos.angle < 67.5:
        motoras.forward()
        servo.backward()
    elif pos.angle > 67.5 and pos.angle < 112.5:
        servo.backward()
    elif pos.angle > 112.5 and pos.angle < 157.5:
        motoras.backward()
        servo.backward()
    elif pos.angle > -157.5 and pos.angle < 157.5:
        motoras.backward(1)
    elif pos.angle > -157.5 and pos.angle < -112.5:
        motoras.backward()
        servo.forward()
    elif pos.angle > -112.5 and pos.angle < -67.5:
        servo.forward()
    elif pos.angle > -67.5 and pos.angle < -22.5:
        motoras.forward()
        servo.forward()        

def stop():
    motoras.stop()
    servo.stop()

bd.when_pressed = move
bd.when_moved = move
bd.when_released = stop

pause()
```
Thank you!

neithere, 2020-12-26:
If you have a two-wheel robot (one motor per wheel), it's easy to use one circle for quite a bunch of movements. I've set a dead zone within 0.2, otherwise `y` is divided into forward/backward and `x` into left/right; then "pure right" (`x` > 0.2, `y` within 0±0.2) means "turn right in place" (left motor forward, right motor backward), but if `y` meaningfully deviates from zero, "forward-ish right" means outer right turn (left wheel forward, right on hold) and "backward-ish right" means inner right turn (left wheel on hold, right wheel backwards). So you can get a lot of precision movements with just a single blue circle; basically everything that is doable with such configuration is covered.