martinohanlon/BlueDot #132: python Errno 98 address in use - how do I clear this

AlistairMcKinnon, 2019-08-28:
My Python program works great with Android Bluedot app.
If I ctrl-C out of python program, it closes everything successfully, and it closes the bluedot screen on Android phone/
If I ctrl-Z out of python program, it does not close everything and leaves bluedot hanging.
When I re start the python program, it says Errno98 Address in use.
How do I free up the address?

AlistairMcKinnon, 2019-08-28:
Thanks for this and thanks for the App!

martinohanlon, 2019-08-28:
Ctrl Z suspends the process, leaving your program running and therefore the BlueDot server as well.

You could trap the suspend event and stop the `BlueDot` using the `.stop()` before quitting. Here is an article about how.

https://stackoverflow.com/questions/49114749/testing-ctrl-z-in-python-script

Or you could use Ctrl C! Is there a reason you are closing using Ctrl Z?

Closing as not an issue with Blue Dot.

closed 2019-08-28 by martinohanlon