martinohanlon/BlueDot #146: no services which can be used with Raspberry Pi

mtyeager, 2020-01-11:
I have gone through the Bluetooth pairing and connection info. The Raspberry Pi pairs to the android phone, but it shows the message "Paired successfully, but this device has no services which can be used with Raspberry Pi." When I launch BlueDot I see the Raspberry Pi, and can click on it, but get the message "unable to connect". I don't see any additional guidance in the readme. Any thoughs?

mtyeager, 2020-01-11:
Nevermind. When I run the python script on the pi, it does connect. Not a problem. Thanks.

closed 2020-01-11 by mtyeager

martinohanlon, 2020-01-11:
Yes. The error is basically saying 'there are no services running on the Pi' (yet) . Your python program is this service your phone will connect too.

mtyeager, 2020-04-22:
Hi Martin,

I have been trying to run the bluedot Bluetooth server in a docker container on RPi 4 (slim-buster).

It keeps failing on import dbus, giving me a modulenotfound error, even though I have RUN apt-get install python3-dbus in the dockerfile.

The Bluetooth server runs fine directly on the RPi hardware, but gives me this error when I try to run it in a docker container.

Any thoughts?

Thanks,

Michael

martinohanlon, 2020-04-22:
@mtyeager Not a clue. Sorry.

My experience of using docker (on a pi or otherwise) is zero.

ukBaz, 2020-04-22:
No idea on docker either but I was curious so I did a search which turned up this:
https://georgik.rocks/how-to-start-d-bus-in-docker-container/

Does that help you @mtyeager ?

mtyeager, 2020-04-22:
Thanks Barry,

My issue is failure on the import dbus…

Do you think that would fail because dbus isn’t running? I thought it would be because it wasn’t installed, or it was installed, but it couldn’t find it…

Thanks so much for your help.

Michael

https://georgik.rocks/how-to-start-d-bus-in-docker-container/

virtadpt, 2020-04-23:
If `import dbus` is failing in your code, you don't have the `dbus` Python module in your container.

mtyeager, 2020-04-23:
Yes, thanks Martin,

Sorry I bothered you with this. It was because I built the image on top of “Slim-buster”.

Which is slim. Not like standard Raspbian…

Once I installed a whole lot of things, it started working properly.

All the best. Stay safe.

Thanks,

Michael