martinohanlon/BlueDot #147: raspberry bluetooth connecte yes then no

ronhud, 2020-02-01:
I have Bluedot on my android phone and it is paired with my raspberry pi.   With the raspberry terminal window open I use Bluedot on my phone and try to connect.  In the terminal window i get the device id and connected yes followed by the id and connected no.  Any suggestion as to a likely reason and fix?

ukBaz, 2020-02-01:
Is this the same as #146 ?

Do you have the python script running on the raspberry pi?
Can you share that script? Or use one of the examples to prove the connection works?

ronhud, 2020-02-01:
Just dicovered an earlier question wit5h instruction to run the wait_for_press script, ok now. Thanks

martinohanlon, 2020-02-01:
Thanks for the speedy response @ukBaz

closed 2020-02-01 by martinohanlon