martinohanlon/BlueDot #149: Need help incorporating a function to robot

Vidura31, 2020-04-26:
Hi I managed to get the Bluedot app successfully working for my robot, however I have been struggling with incorporating a code (please see below) from my obstacle avoiding robot. Please note that I'm still new to programming. Basically, I created a new function to put the robot in to autonomous mode, however it gets stuck in the while loop within the function once its called. It goes back to the function and I tried to break out of it, but I have been unsuccessful. I want to break out of the loop once I swipe or press a button again. Can you please tell me how to achieve this?

GPIO.setmode(GPIO.BCM)
bd = BlueDot()  

robot = Robot(left=(4, 17), right=(27, 22))

def move(pos):
        if pos.top:
             robot.forward()
             print("forward")
        elif pos.bottom:
             robot.backward()
             print("backward")
        elif pos.left:
             robot.left()
             print("left")
        elif pos.middle:
             robot.right()
             print("right")

def auto_drive():
        while True:
                TRIG= 18
                ECHO = 23
                GPIO.setup(TRIG, GPIO.OUT)
                GPIO.setup(ECHO, GPIO.IN)

                GPIO.output(TRIG, True)
                time.sleep(0.0001)
                GPIO.output(TRIG, False)

                while GPIO.input(ECHO) == False:
                    start = time.time()

                while GPIO.input(ECHO) == True:
                    end = time.time()

                signal_time = end-start
                print(signal_time)

            #speed of sound = 343m/s = 34300cm/s
            #speed = d*T (T/2 for reflection and 343*100 for cm/s)
                distance = (signal_time/2)*34300

                print("Distance: {} cm".format(distance))
                time.sleep(0.1)
                
                if distance < 30:
                     robot.stop()
                     time.sleep(1)
                     robot.backward()             
                     time.sleep(2)
                     robot.right()
                     time.sleep(1)
                     robot.forward()
                     time.sleep(1)
                else:
                     robot.forward()
def stop():
        robot.stop()
     
bd.when_pressed = move
bd.when_moved = move
bd.when_released = stop
bd.when_swiped = auto_drive

pause()

martinohanlon, 2020-04-26:
I edited your issue to format the code.

There are numerous ways of solving your problem. Ultimately you need to get your auto drive function running in a thread (or in the background).

Using the `set_when_swiped` function you can get your auto_drive function to run in the background.

You also need to set a condition in your while loop so it does continue forever. 

I had a hack about and came up with this which uses a global variable which is set to False when bluedot in double pressed. It is not tested tho.

```python
GPIO.setmode(GPIO.BCM)
bd = BlueDot()  

robot = Robot(left=(4, 17), right=(27, 22))
driving = False

def move(pos):
        if pos.top:
             robot.forward()
             print("forward")
        elif pos.bottom:
             robot.backward()
             print("backward")
        elif pos.left:
             robot.left()
             print("left")
        elif pos.middle:
             robot.right()
             print("right")

def stop_auto_drive():
    global driving
    driving = False

def auto_drive():
        global driving
        driving = True
        while driving:
                TRIG= 18
                ECHO = 23
                GPIO.setup(TRIG, GPIO.OUT)
                GPIO.setup(ECHO, GPIO.IN)

                GPIO.output(TRIG, True)
                time.sleep(0.0001)
                GPIO.output(TRIG, False)

                while GPIO.input(ECHO) == False:
                    start = time.time()

                while GPIO.input(ECHO) == True:
                    end = time.time()

                signal_time = end-start
                print(signal_time)

            #speed of sound = 343m/s = 34300cm/s
            #speed = d*T (T/2 for reflection and 343*100 for cm/s)
                distance = (signal_time/2)*34300

                print("Distance: {} cm".format(distance))
                time.sleep(0.1)
                
                if distance < 30:
                     robot.stop()
                     time.sleep(1)
                     robot.backward()             
                     time.sleep(2)
                     robot.right()
                     time.sleep(1)
                     robot.forward()
                     time.sleep(1)
                else:
                     robot.forward()
def stop():
        robot.stop()

bd.when_pressed = move
bd.when_moved = move
bd.when_released = stop

bd.set_when_swiped(auto_drive, background=True)
bd.when_double_pressed = stop_auto_drive

pause()
```

You will find it easier to raise questions like this on the raspberry pi forum, https://www.raspberrypi.org/forums/, there are much more people available there who can help. I am the only person who monitors these issues.

As this is not an issue with BlueDot, I am going to close it.

closed 2020-04-26 by martinohanlon

Vidura31, 2020-04-27:
Hi

Many thanks for your message and clarifying this. I will try this code today. I tried to work this out for hours.

Apologies for posting the question in the wrong place.

Kind regards

Vidura