martinohanlon/BlueDot #150: Multiple comm: Bluetooth Comm API

orionJag, 2020-05-10:
**Describe the feature you'd like**
Currently using Bluetooth Comm API and I was wondering if multiple clients can send request to the server. I am using "bluedot.btcomm.BluetoothServer" and "bluedot.btcomm.BluetoothClient" API and when more than 1 clients makes a request, the subsequent client just hangs.  

Not sure if this is a feature request, guess is I  am missing some additional configuration to allow multiple connectivity to the Server.

Appreciate the help and support. 

Thanks

martinohanlon, 2020-05-11:
Bluetooth is a 1 to 1 connection. A server can only accept a single connection.

You could start multiple servers on different ports that way you could host multiple connections.

closed 2020-05-11 by martinohanlon

martinohanlon, 2020-05-11:
For clarification btcomm is a 1 to 1 connection. I think it's technically possible to have multiple rfcomm connections with BT, although no idea if it is supported by pythons socket layer.

ukBaz, 2020-05-11:
From https://docs.python.org/3/library/socket.html:

> BTPROTO_RFCOMM accepts (bdaddr, channel) where bdaddr is the Bluetooth address as a string and channel is an integer.

So I think you would have to define each server to listen on a different channel.

martinohanlon, 2020-05-11:
Thank you @ukBaz . In which case. @orionJag as I already said, you would have to use different ports.

orionJag, 2020-05-11:
Martin & @ukBaz 

           Thanks for such a quick response and clarification.  I will check out using different ports. 

           Appreciate all your support to the community.

Thanks