martinohanlon/BlueDot #152: Unable to connect with my Pi

GireeshHarish, 2020-07-14:
Hello,
I was trying to use BlueDot for controlling the GPIO pins of my Raspberry Pi 3B but unfortunately, I am not able to connect with my device. The pairing was successful but I am not able to connect my Pi and device using Bluetooth. Can you please help me with this issue. I will attach the screenshot of the error message also here.
[image]

martinohanlon, 2020-07-14:
It looks like you are trying to connect to the Blue Dot app by using the bluetooth menu on your Raspberry Pi - is that right?

The Blue Dot app connects to your Raspberry Pi.

It works like this:
- pair your phone and raspberry pi
- write your Python program 
- RUN your Python program
- when it is running
- open the Blue Dot app on your phone
- select your Raspberry Pi from the list of devices on the Blue Dot app on your phone

This is documented in the getting started guide. https://bluedot.readthedocs.io/en/latest/gettingstarted.html

GireeshHarish, 2020-07-15:
Thank you very much. It worked like a charm.

cchokiGyeltshen, 2024-05-06:
sudo pip3 install  bluedot
on entering this command I am getting the respond as instead try apt install python3-xyz
I did this and somehow I managed to install python3-xyz. However, it prompts 
E- unable to locate package python3-xyz

How can I resolve this issue?