martinohanlon/BlueDot #159: Be able to add text as well as buttons

zax71, 2020-12-12:
I would like a way to add text to a blue dot grid, maybe having it take up multiple rows/columns 

If this is already a feature i must of missed it in the docs!

martinohanlon, 2020-12-26:
This is not currently something you can do with bluedot.

I will leave it as a feature request for the time being but it's unlikely to be a priority for me.

IVIJL, 2021-04-26:
This is only thing which is missing there, to label buttons if I got like 10 buttons on page I dont remember wich does what, so basic text label inside would be perfect.

JediNizar, 2021-09-09:
Yep it's something I would apperciate too

McFutzel, 2022-10-26:
just to bring it back up... i´d be interested too

rvmey, 2023-02-04:
Same here.  Text on each dot would be super helpful.

MacanPN, 2023-09-20:
Same here! Sounds like a very simple thing to implement and would help to build more complex UIs.