martinohanlon/BlueDot #160: 'Unable to Connect' Error

omarhassan91, 2020-12-25:
Hello!

I've downloaded the BlueDot application on my android phone with the intention of controlling some modules I have connected to the GPIO pins on a Raspberry Pi Zero WH (which has built in Bluetooth support). I am able to successfully pair my android phone to the raspberry pi, however when I try to connect to the Pi through the BlueDot Application, I get an error saying 'Unable to Connect'. 

I also tried running the following code on the Pi before connecting:

```
from bluedot import BlueDot
bd = BlueDot()
bd.wait_for_press()
print("You pressed the blue dot!")
```

I ran that code, waited for the prompt to say 'Waiting for Connection', and then I tried connecting to the Pi in the BlueDot app. Same 'Unable to Connect' error...

Has anyone encountered this issue, or know if there's a step that I'm missing?

Thanks!

martinohanlon, 2020-12-25:
To confirm.

Your Pi and phone need to be paired together.
Your Pi needs to be running the Python program.
You should then be able to open the BlueDot app and connect.

If this fails I would unpair your phone and Pi. Delete the pi from you phone and phone from your Pi.

Follow these instructions https://bluedot.readthedocs.io/en/latest/gettingstarted.html to pair your phone and Pi, run your python program and connect.

You got this...

omarhassan91, 2020-12-25:
Hello, 

I tried all of this prior to my posting. I've always made sure that my Pi is paired successfully to my phone via Bluetooth, and I run the above Python code before I even try to connect to the Pi through the app. Could this be an issue with the Pi I'm using?

martinohanlon, 2020-12-26:
Maybe... If you are confident everything is setup ok. I would be doing things like trying a different pi, trying a different phone, reimagining the Pi's sd card and trying again.

Wroxham, 2020-12-26:
Hello Martin,
Like Omar, I'm having the same problem. I'm using a two week old Raspberry Pi 400, all software updated - its practically new out of the box. I've connected two separate phones via bluetooth and sucessfully uploaded and downloaded test files over bluetooth. With Python running one of your codes and 'Waiting for Connection' onscreeen, the Blue Dot app just fails to connect. I've used two separate android phones now. I've gone through the loop several times of forgetting devices (on the Pi and on each phone) and starting the conection process again from scratch, all to no avail.  The Blue Dot app subbornly refuses to connect. Has anything changed to stop new downloads of the app working? Any advice would be appreciated.

ukBaz, 2020-12-26:
There are a couple of places you can look for more debug information with Bluetooth.

This simplest is the command `service bluetooth status` which will have a section at the bottom that might be reporting errors.

This is fairly similar to doing `journalctl -r -u bluetooth` but will reveal messages from further back in time if that is important.

If that doesn't yield any clues, then having `bluetoothctl` open in another terminal might report some additional information.

For the lowest level of reporting then run `sudo btmon` in another terminal when you are trying to connect. This is very low level and verbose information but often will give an error message that gives a clue as to what might be going wrong.

martinohanlon, 2020-12-26:
I will test with the latest versions of Raspberry Pi OS (there was a recent release). It's possible an issue has been introduced.

martinohanlon, 2020-12-26:
Ok. I have tested with the latest version and I am seeing the same issue.

Running `sudo btmon` shows the following when trying to start up the `BlueDot` server.

ukBaz, 2020-12-26:
Did you forget to include the `btmon` output?

martinohanlon, 2020-12-26:
@ukBaz - pressed enter too soon - comment updated.

ukBaz, 2020-12-26:
@martinohanlon I currently have the following on a Pi Model 3B  V1.2 :
 
```none
pi@SensePi:~ $ dpkg -l | grep bluez
ii  bluez                                5.50-1.2~deb10u1+rpt2               armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt6                          all          Firmware for Bluetooth devices
```
And I am seeing it too.  I see there is an `bluez-firmware all 1.2-4+rpt7` so I'm doing an update and will retest....

martinohanlon, 2020-12-26:
Hmmmm...  Something interesing. I think bluedot and pulseaudio use the same UUID for setting up SPP.

If I change the UUID used by bluedot `uuid = "00001101-0000-1000-8000-00805f9b34fb"` to something different and run the BlueDot server I dont see the same error reported in btmon. 

I vaguely remember that this UUID is the one used in examples of how to setup SPP. Could me and the pulseaudio developers copied the same examples? 😬 

To test this I will need to change the Android app...

martinohanlon, 2020-12-26:
Right... I need to do some more investigation. If I force the android app and the BlueDot to use a different port e.g. 11. It works (regardless of what UUID use).

ukBaz, 2020-12-26:
I think changing the port is the way to go rather than the UUID. 

A Serial Port connection should have the UUID of `00001101-0000-1000-8000-00805f9b34fb`
https://www.bluetooth.com/specifications/assigned-numbers/service-discovery/

omarhassan91, 2020-12-26:
Hey, 

This is great! I tried doing `pulseaudio -k` and it worked! I was able to get the phone to connect to the Pi and the Blue Dot worked :)

Now a separate question, I know this goes a little back to basics, but is there a guide or some documentation I can follow in regards to controlling certain GPIO pins with the app?

omarhassan91, 2020-12-26:
Sorry don't mean controlling GPIO pins, I mean pulling data from specific GPIO pins

ukBaz, 2020-12-27:
@martinohanlon I've done a little bit of digging around on the SPP/RFCOMM setup.

Seems like a recommended approach is that the channel/port isn't hard coded and at connection time it is somehow negotiated. [[1](https://www.btframework.com/rfcomm.htm#channel)]

Seems like the [profile_uuid](https://gist.github.com/ukBaz/217875c83c2535d22a16ba38fc8f2a91#file-spp-profile-L63) can be different from the [service_uuid](https://gist.github.com/ukBaz/217875c83c2535d22a16ba38fc8f2a91#file-spp-profile-L69)
So I wonder if you can have a custom UUID for the BlueDot profile but have the service as `1101` still?

Also, I've worked out that there isn't a need for the [SDP record to be manually](https://github.com/martinohanlon/BlueDot/blob/489eb61479599b97ebf7025ea70c8ab1c8b36833/bluedot/utils.py#L101) enter as [BlueZ has some sensible defaults](https://github.com/bluez/bluez/blob/c96d5ec7aafc8fd988f6f19b896115853b951d91/src/profile.c#L2009)

(After doing `pulseaudio -k`) The following worked for me with the BlueDot Android App: https://gist.github.com/ukBaz/217875c83c2535d22a16ba38fc8f2a91#file-spp-profile

Would this be a good time to talk about moving to BLE?  :-)

martinohanlon, 2020-12-27:
Ok, so a little more investigation. 

To recap how BlueDot works today:

- BlueDot registers a SP profile with the "standard" UUID on whatever port is specified via the `BlueDot` class.
- When the Android app is using "Auto port discovery" (you can change it in the app settings) it uses `[createRfcommSocketToServiceRecord](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#createRfcommSocketToServiceRecord(java.util.UUID))` to "automatically" connect the correct port

This is what I "think" is happening is that pulseaudio is also registering a SP profile using the same UUID [I think] on port 3. Android then tries to connect to the pulseaudio SP profile.

If you change the BlueDot settings to turn off "auto port discovery" and specify a specific port, the android app will use `createRfcommSocket` to invoke a connection direct to a specific port and it works.

@omarhassan91 @Wroxham - The easiest solution to your problem right now will be to open up the setting in BlueDot (the 3 dots in the top right) turn off "Auto port discovery" and select port 1.

martinohanlon, 2020-12-27:
@ukBaz 

> Seems like the profile_uuid can be different from the service_uuid
> So I wonder if you can have a custom UUID for the BlueDot profile but have the service as 1101 still?

Good question... Only one way to find out I suppose. I wonder how Android finds the service using `createRfcommSocketToServiceRecord` - documentation seems a little sketchy

> (After doing pulseaudio -k) The following worked for me with the BlueDot Android App:
> https://gist.github.com/ukBaz/217875c83c2535d22a16ba38fc8f2a91#file-spp-profile

I will have a go and see if I can get this working.

> Would this be a good time to talk about moving to BLE? :-)
It always been on the plans. I need to find time to create a Python and Android POC. See if I can get them talking without having to do any crazy setup. When I have a solution for this issue I will create a new issue to explore.

martinohanlon, 2020-12-27:
> Also, I've worked out that there isn't a need for the SDP record to be manually enter as BlueZ has some sensible defaults

I couldn't find a way of changing the channel when using the BlueZ defaults.

martinohanlon, 2020-12-27:
> I couldn't find a way of changing the channel when using the BlueZ defaults.

Ignore that you set the port, when you open the socket, not when you create the SDP record.

martinohanlon, 2020-12-27:
When pulseaudio is running I cannot get Android to create an rfcomm socket using the service record. e.g. using createRfcommSocketToServiceRecord. I have tried changing the service and profile UUIDs all with the same result.

I am thinking the easiest route is to modify the android app to direct connect to port 1 (the default) unless someone changes the port in the settings. It simple and probably reliable. I need to do some testing though.

Wroxham, 2020-12-27:
Many thanks for providing support on this.  I now have the app working with auto port discovery switched off. I've connected successfully through ports 1 and 2 and even connected two mobiles simultaneously running python code for bd1 and bd2. Auto port discovery works fine with pulseaudio -k.  I know you already know all this but I'm just confirming that my experience with the app concurs with what you are saying.
The app is absolutely brilliant :-) Thank you.

martinohanlon, 2020-12-27:
Thank you for the confirmation @Wroxham

ukBaz, 2020-12-28:
The [range for the channel number](https://people.csail.mit.edu/albert/bluez-intro/x148.html#port-numbers) is 1 - 30. I would be tempted to go higher in the range as 1 & 3 seem to be commonly used by various applicaitons.

martinohanlon, 2020-12-28:
Thanks. Yes, the android app lets you pick a channel between 1 and 30. 

BlueDot has always used port 1 as the default. For the moment I will leave it as is as I dont want to introduce more change than I have to right now [if I change the default port I would also have to release a new bluedot Python library, which would create an incompatability and those who didnt have the very latest version of both would experience problems]

martinohanlon, 2020-12-28:
Released a new version of the Blue Dot android client which does away with "auto port discovery" in favour of connecting directly to a port. It has been released on the Google Play store, is currently awaiting review and will roll out automatically after that.

The new APK is here - https://github.com/martinohanlon/BlueDot/blob/master/clients/android/app/release/app-release.apk

martinohanlon, 2020-12-30:
New version is being rolled out. Closing this issue. Thanks for reporting :)

closed 2020-12-30 by martinohanlon