martinohanlon/BlueDot #166: auto run from Pi's Shell

YXGuan, 2021-01-26:
Hello Martin,

First of all, thank you for the amazing software!

My question: is there a way to pair Pi to my phone and run some Bluedot codes automatically every time Pi boots up?

The issue I have: it gives me an error when I tried to run my program from Pi's **shell**:
`No module named blue dot` 

Here's a screenshot of my error:
[image]
[screenshot](https://drive.google.com/file/d/1d3QzNu18WIiqqoCAwoWw1xXAC3gJKYSH/view?usp=sharing)

Could you help me out?

Thank you for your time.

ukBaz, 2021-01-26:
I noticed that you did a `pip3` (i.e. install it into your python3 setup) but then used the python command which by default will run python2.

Try using the `python3` command instead.

martinohanlon, 2021-01-26:
I would encourage you to head to the Raspberry Pi Forum raspberrypi.org/forum you will get a faster response and there is a much larger group of people who can help you.

A few points:
- as @ukBaz points out, you are trying to run your program using Python 2 e.g. `python` - use `python3` instead e.g. `python3 /home/pi/path_to_program.py`
- you only need to pair your phone once, you wont need to do it everytime the pi boots
- the simplest way to get your program to boot is probably by adding the command to `rc.local` - if you search Im sure you will find a tutorial.

I am closing the issue as it isnt a bug with BlueDot.

closed 2021-01-26 by martinohanlon

ukBaz, 2021-01-26:
Apologies in advance to Martin for posting this, but there are various advantages to using `systemd` over `rc.local`. I find the following a useful reference:
https://blog.usedbytes.com/2019/11/run-at-startup-without-rc.local/

martinohanlon, 2021-01-26:
Many many advantages of using systemd. The key disadvantage being that its a lot more complicated.

YXGuan, 2021-01-26:
@martinohanlon @ukBaz 
Thank you all, Martin and Barry. Like you guys suggested: python3 fix the problem.
Thanks again, it's really helpful.
Have a wonderful day!

YXGuan, 2021-01-26:
Keep up the good work!