martinohanlon/BlueDot #168: Documentation: requirement for python-dbus (dbus-python) no longer directly mentioned in docs

ricksorensen, 2021-02-22:
I was using a Raspberry Pi Zero W .. with latest updates, and a python3 virtual environment.   As su I ran the dpad,.py example app, and received an error indicating the dbus module was not available:

Traceback (most recent call last):
  File "dpad.py", line 1, in <module>
    from bluedot import BlueDot
  File "/home/pi/rjsvenv/bdot/lib/python3.7/site-packages/bluedot/__init__.py", line 1, in <module>
    from .dot import BlueDot, BlueDotButton
  File "/home/pi/rjsvenv/bdot/lib/python3.7/site-packages/bluedot/dot.py", line 9, in <module>
    from .btcomm import BluetoothServer
  File "/home/pi/rjsvenv/bdot/lib/python3.7/site-packages/bluedot/btcomm.py", line 7, in <module>
    from .utils import (
  File "/home/pi/rjsvenv/bdot/lib/python3.7/site-packages/bluedot/utils.py", line 3, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'

manually nstalling the dbus-python module resolves the dependency

Server started B8:27:EB:5C:D0:AD
Waiting for connection

Steps to reproduce the behavior:
 2064  python3 -m venv rjsvenv/bdot
 2065  cd rjsvenv/bdot
 2066  . bin/activate
 2067  pip install -U pip wheels
 2068  pip install -U pip wheel
 2069  cp -p ../bluetooth/dpad.py  .   # get example/dpad.py in work directory
 2070  pip install bluedot
 2071  sudo /home/pi/rjsvenv/bdot/bin/python dpad.py    # generates error
 2074  pip install dbus-python
 2075  sudo /home/pi/rjsvenv/bdot/bin/python dpad.py   # error resolved

**System (please complete the following information):**
- HW: Raspberry Pi Zero W 
 - OS: Raspian
 - Version: Buster/Linux  Linux raspberrypizx 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l GNU/Linux

**Additional context**
Earlier documentation ([https://bluedot.readthedocs.io/en/master/gettingstarted.html]) suggests apt/dpkg to install dbus for python system wide.  
Current documentation   [https://bluedot.readthedocs.io/en/latest/gettingstarted.html] does not mention dbus.

I used sudo to get access to the bluetooth components on the Pi ...  will add user to the permissions later.

I am not sure how pip determines which dependencies to bring in automatically, but it is not finding anything in the bluedot module (utils,py imports dbus)

ukBaz, 2021-02-23:
Looking at the steps you have outlined and looking at the install instructions in the Blue Dot documentation, I would expect the issue to be that you have installed Blue Dot into a Python Virtual Environment while the documentations installs Blue Dot into the system Python.
`dbus-python` is already installed into the system Python and so its installation doesn't need to be documented. While because you have created a venv without the `--system-site-packages` flag, you need to install it within your venv.

I would also question whether you needed to use `sudo`. Without `sudo` you wouldn't need to put the path to the `venv` Python as it would be the default Python that would be run.

ricksorensen, 2021-02-23:
Thanks for the response .. yes indeed I am using a python virtual environment without `--system-site-packages` which is my usual development process.   I also do not have `dbus-python` installed in my default python (neither 2 nor 3)- I keep that environment pretty sparse.  I only use `sudo` for access to the dbus .. until I update that access for my standard user.  
As I tried to indicate this looks like a documentation issue - the gettingstarted section for version [1.0.4](https://bluedot.readthedocs.io/en/master/gettingstarted.html) mentions the install of dbus-python into the system python, but the latest [2.0.0](https://bluedot.readthedocs.io/en/master/gettingstarted.html) does not.

closed 2021-02-23 by ricksorensen

ukBaz, 2021-02-23:
Interested to hear that you don't have `dbus` by default. I thought it was one of the default packages on Raspberry Pi.

Are you running with a full desktop or are you running the lite version?

With system Python it will not be in the `pip3 list` as it is a Linux distribution package. Maybe this will help explain:

```shell
pi@SensePi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@SensePi:~ $ uname -a
Linux SensePi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
pi@SensePi:~ $ pip3 list | grep dbus
pi@SensePi:~ $ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dbus
>>> dbus.__file__
'/usr/lib/python3/dist-packages/dbus/__init__.py'
>>> exit()
pi@SensePi:~ $ cd /tmp
pi@SensePi:/tmp $ python3 -m venv venv
pi@SensePi:/tmp $ . venv/bin/activate
(venv) pi@SensePi:/tmp $ python
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'dbus'
>>> 
```

ricksorensen, 2021-02-23:
I am running headless- no GUI.  I don't think it is the official 'lite' install, but I have no X libs or apps installed.  

Besides looking through the package lists and pip installs at system level (not virtualenv) I tried command 
`pi@raspberrypizx:~ $ which python3
/usr/bin/python3
pi@raspberrypizx:~ $ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'dbus'
>>> exit()
pi@raspberrypizx:~ $
`