martinohanlon/BlueDot #170: Unablet to connect

MDiPas, 2021-09-02:
Anyone know how solve the problem unable to connect?

Model: raspberry Pi 4 B 8 gb Ram
I tried to open before App and then run the script
I tried to run before script and after the app
I tried to unistall the os and reinstall lite version / version with the desktop / version with desktop and suggest software
I tried to change android device

Anyone can suggest my other solution?
[image]
[image]
[image]

martinohanlon, 2021-09-02:
You have a different device (either raspberry pi or bluetooth adapter) paired with your mobile phone than you are running your python program on.

You can see the mac addresses which you are trying to connect to and that the server are running on are different.

[image]

[image]

I would try removing all the pairings you have made (on your phone and raspberry pi) and starting again.

MDiPas, 2021-09-02:
what is that mac? I have try to block but the system doesn't know it
[image]

MDiPas, 2021-09-02:
[image]

MDiPas, 2021-09-04:
I have found Mac :  ether e4:5f:01:38:e9:3a
[image]
[image]

Now, How i solve it?

ukBaz, 2021-09-04:
It makes no sense that it is reporting the `eth0` interface address. Can you share the code that you are using?

Or run the following example to see what that reports?
https://github.com/martinohanlon/BlueDot/blob/master/examples/adapter_details.py 

It would also be interesting to use the following technique in the Linux terminal to verify a different way what the Bluetooth mac address is:
```
$ busctl get-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Address

```

MDiPas, 2021-09-04:
[image]

ukBaz, 2021-09-04:
The fact that the `busctl` command is giving the "wrong" answer suggests it might be an issue more fundamental than the BlueDot library. 
Does `service bluetooth status` show errors?
Does `sudo service bluetooth restart` work?

When running your BlueDot script have separate terminals open with the following running to get more debug information:
`bluetootctl`
`journalctl -f -u bluetooth`
`sudo busctl monitor org.bluez`
`sudo btmon`

MDiPas, 2021-09-04:
[image]

martinohanlon, 2021-09-07:
I have just tried to recreate this [spoiler I cant] using:
- raspberry pi 4
- latest Raspberry Pi OS release (Linux raspberrypi 5.10.17-v7l+)

`hcitool` and `bluedot` return the mac address for `hci0`.

[image]

I have no idea why bluez would return the mac address of `eth0`. I am at a loss. 

@ukBaz any other insight you have would be appreciated. 

I will leave the issue open in case other users experience the same issue, but right now I dont know what to suggest.

ukBaz, 2021-09-07:
Sorry @martinohanlon at a bit of a loss as to what to suggest with this. As I said further up the thread, the fact that `busctl` is giving the wrong value for the Bluetooth adapter suggests it is at a lower level than Bluedot. The only thing I can suggest to @MDiPas is to go through the journal for last boot (`journalctl --boot`) and see if there are any error messages that give a clue where to look.

There has been some playing around with how the `bdaddr` is derived so it might be worth checking that the latest RPi Bluetooth firmware is loaded. As an example:
```
pi@SensePi:~ $ dpkg -l | grep bluetooth
ii  lxplug-bluetooth                     0.21                                    armhf        Bluetooth plugin for lxpanel
ii  pi-bluetooth                         0.1.17                                  all          Raspberry Pi 3 bluetooth
ii  pulseaudio-module-bluetooth          12.2-4+deb10u1+rpi3                     armhf        Bluetooth module for PulseAudio sound server
```
`pi-bluetooth` is the key one to check.  Looking at https://github.com/RPi-Distro/pi-bluetooth, `0.1.17` seems to be the latest.

martinohanlon, 2023-01-26:
Closing due to lack of activity.

closed 2023-01-26 by martinohanlon

Laxin1234, 2024-05-05:
unable to install bluedot on raspberry pi 5 by using sudo pip3 install bluedot
it shows:error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

even if I us sudo apt install or sudo apt get it shows Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package bluedot
what should I do?