martinohanlon/BlueDot #172: Web Bluetooth support and GATT

clemos, 2021-11-12:
Hi there :wave:

I love BlueDot, and wanted to contribute a web client using [Web Bluetooth](https://developer.mozilla.org/en-US/docs/Web/API/Web_Bluetooth_API).

I've made a few tests (I'm not particularily an expert in Bluetooth in general), and it seems it is not possible to connect from Web Bluetooth to a service that is not advertised by the device's GATT server, and it seems BlueDot is not visible to Web Bluetooth for that reason.

Can you confirm my conclusions ? And if so, do you have any idea how one could support this feature ?

Thanks

ukBaz, 2021-11-12:
@clemos , You are correct: Web Bluetooth only supports BLE. Which is incompatible with BlueDot as it uses the Serial Port Profile from Bluetooth Classic.

Various hardware has implemented the Nordic UART Service (or similar) as a way emulate UART/Serial Port over BLE but that would be a major change to how BlueDot works.

Laxin1234, 2024-05-05:
unable to install bluedot on raspberry pi 5 by using sudo pip3 install bluedot
it shows:error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

even if I us sudo apt install or sudo apt get it shows Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package bluedot
what should I do?

virtadpt, 2024-05-05:
Installing a Python module system-wide with Pip is a bad idea, and it's warning you not to do that.  It's safer and more useful to install it into a virtual environment.

```
user@you: ~/ $ python3 -mvenv your_project_here
user@you: ~/ $ . your_project_here/bin/activate
(your_project_here) user@you: ~/ $ pip install bluedot
```

Laxin1234, 2024-05-07:
when I put this code
 from bluedot import BlueDot
from gpiozero import LED

bd = BlueDot()
led = LED(16)

while True:
    bd.wait_for_press()
    led.on()
    bd.wait_for_release()
    led.off()

in shell it shows
>>> %FastDebug bluetooth1.py
Traceback (most recent call last):
  File "/home/laxin/python/bluetooth1.py", line 1
    from bluedot import BlueDot
IndentationError: unexpected indent
even after I did step told by virtadpt
laxin@raspberrypi:~ $  python3 -mvenv your_project_here
laxin@raspberrypi:~ $  . your_project_here/bin/activate
(your_project_here) laxin@raspberrypi:~ $  pip3 install bluedot
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: bluedot in ./your_project_here/lib/python3.11/site-packages (2.0.0)
please tell what to do next

virtadpt, 2024-05-07:
`IndentationError: unexpected indent`

Your code's not right - Python is whitespace sensitive.  Looking at your code (pasted above), it should look like this:

```
from bluedot import BlueDot
from gpiozero import LED

bd = BlueDot()
led = LED(16)

while True:
    bd.wait_for_press()
    led.on()
    bd.wait_for_release()
    led.off()
```

Here, take a look at this - it's some code I wrote a while back that uses Bluedot.  It mighht help: https://github.com/virtadpt/pwnagotchi-bt-power

clach04, 2024-12-29:
@virtadpt thanks for posting! I'm probably going to steal your idea/approach in https://github.com/virtadpt/pwnagotchi-bt-power for shutting down a music radio device I built, very neat idea!