martinohanlon/BlueDot #177: not possible to check when bluetooth connexion in on

Georges1944, 2022-04-30:
When I launch the script simpledot.py :

from bluedot import BlueDot
print(1)
bd = BlueDot()
print(2)
bd.wait_for_press()
print("You pressed the blue dot!")

I get following result.
--------------------------------------------------
pi@raspitest1:$ simpledot.py
1
Server started E4:5F:01:7F:2C:99
Waiting for connection
2
Client connected 58:A6:39:F3:7B:59
You pressed the blue dot!

---------------------------------------------------------
Is there a way to know that the client is connected ? 

Thanks for help, and congratulation for this great work !
Georges

ukBaz, 2022-04-30:
The following allows for a function to be called when a Blue Dot connects

https://bluedot.readthedocs.io/en/latest/dotapi.html#bluedot.BlueDot.set_when_client_connects

Or there is a property with the status:
https://bluedot.readthedocs.io/en/latest/dotapi.html#bluedot.BlueDot.is_connected

Is that what you are looking for?

martinohanlon, 2022-05-01:
What @ukBaz said.

You can also use `.when_client_connects`.

```python
def a_client_connected():
    print ('a client connected')

bd.when_client_connects = a_client_connected
```

Closing as function already exists.

closed 2022-05-01 by martinohanlon