martinohanlon/BlueDot #178: Send connection request from BluetoothServer side as well

ada-lovelace-f5, 2022-06-27:
**Describe the feature you'd like** 
I was working on building an emulator and the BlueDot project has been immensely useful and worked right out flawlessly. 

However, I faced a situation where my emulator was running the BluetoothServer and needed to send an SPP connection to the client Android device. I looked around in the code as well to see if something could be done in similar lines of BluetoothClinet.connect(). Unfortunately, I failed. 

I wanted to ask the community and the devs if this would be a feature/enhancement you would find interesting/useful to be included in the project. 

Looking forward to your comments. 

Thanks team, you are life saver for me.

martinohanlon, 2022-06-27:
A client socket (which is what the android app is) cannot accept and
incoming connection. This isn't a limitation with the BluetoothClient class
it is just how sockets work.

To accept a connection the android app would have to be running a server
socket and be waiting for a connection.

ada-lovelace-f5, 2022-06-28:
@martinohanlon Thank you for your insightful comments. I have understood the way it works much better thanks to you.
I am closing this issue with your comments as closing remarks.

Thanks for all the amazing work you have done for the community.

closed 2022-06-28 by ada-lovelace-f5