martinohanlon/BlueDot #181: Bluedot android app crashes on pi connection

Merisiel0, 2023-01-01:
System:
Pi os: Rasbian
Android phone: Oneplus 8

Using the mydot.py program:
from bluedot import BlueDot
bd = BlueDot()
bd.wait_for_press()
print("You pressed the blue dot!")

- At first the app crashed on launch. Solution was to give it the "Nearby devices" permission.
- Now it crashes when I click on my pi's address.
- If I disconnect my pi (while remaining paired) and then click on it's address on the app, then I get a split second of a screen with a message similar to "Connecting to pi", then the app crashes. This trick won't work a second time unless I connect and disconnect again.
[image]
[image]

martinohanlon, 2023-01-01:
I cant reproduce the error. I have just done a run on the latest version of Raspberry Pi OS, BlueDot python library and the BlueDot android app.

Can you confirm what version of Android you are running? I dont have an Android 12 device to test.

Merisiel0, 2023-01-01:
My android version is 12.

I also have an old iPad which I don't know the version. I can try Blue Dot
with it later today

Merisiel0, 2023-01-01:
Here are my system's versions:
Rasbian: Release 11
Android: 12
Bluedot: 2

martinohanlon, 2023-01-02:
Thank you . I think this is an issue directly related to android 12. I did a release a few days ago to uplift the API version (in order to stay in the play store.

> If there is a way to test remotely, I am willing to let you use my phone.

I dont believe so. If you have the ability to install android studio, run the source on your phone and let me have the results that would be a big help, but its a big ask.

martinohanlon, 2023-01-02:
I am pretty sure this is because I wasnt requesting bluetooth permissions correctly when using Android 12. (Permission changes were a big part of 12).

I have built a new version which runs on the emulator (but there is no way to fully test bluetooth applications on the emulator) - could you download the apk (android installation) and test it for me? Its here - https://github.com/martinohanlon/BlueDot/blob/dev/clients/android/app/release/app-release.apk

You will probably have to uninstall BlueDot first before you can install the APK. There is a good chance android will ask if you want to install from an "unknown source" (or similar).

Boeeerb, 2023-01-02:
Just jumping on this to help out with debugging, using Android 13 (pretty much the same as 12 in permissions department), Android Studio debugging gives the following when clicking on a previously paired (outside of Blue Dot);

```
2023-01-02 11:26:58.840 1902-2267/? E/bluetooth: packages/modules/Bluetooth/system/gd/os/linux_generic/files.cc:202 FileCreatedTime: unable to read '/data/misc/bluetooth/logs/btsnooz_hci.log' file metadata, error: No such file or directory
2023-01-02 11:26:59.840 1475-1731/? E/ContextHubClientManager: Cannot send message to unregistered client (host endpoint ID = -28638)
2023-01-02 11:27:05.145 24860-24903/? E/AndroidRuntime: FATAL EXCEPTION: ConnectThreadSecure1
    Process: com.stuffaboutcode.bluedot, PID: 24860
    java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@690055b1: AdapterService cancelDiscovery
        at com.android.bluetooth.Utils.checkPermissionForDataDelivery(Utils.java:482)
        at com.android.bluetooth.Utils.checkScanPermissionForDataDelivery(Utils.java:541)
        at com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder.cancelDiscovery(AdapterService.java:2038)
        at com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder.cancelDiscovery(AdapterService.java:2030)
        at android.bluetooth.IBluetooth$Stub.onTransact(IBluetooth.java:594)
        at android.os.Binder.execTransactInternal(Binder.java:1285)
        at android.os.Binder.execTransact(Binder.java:1244)
```

Main takeaway I can see is not requesting the correct permission - on first install it asks for location permission;

`Need android.permission.BLUETOOTH_SCAN permission `

martinohanlon, 2023-01-02:
> on first install it asks for location permission;
I think it will do that if you ask for any bluetooth permission.

I thought the app only needed BLUETOOTH_CONNECT, but it must need BLUETOOTH_SCAN as well. Im not sure why but the rules are pretty opaque as to what are needed for which operation.

@Boeeerb thanks so much

ukBaz, 2023-01-02:
Sounds like you are on top of this @martinohanlon .

I'll put this link here as I think it explains what permissions are needed for what:
https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

`BLUETOOTH_SCAN` appears to be needed for querying the list of paired devices which I guess is what Bluedot does:

https://developer.android.com/guide/topics/connectivity/bluetooth/find-bluetooth-devices

> Using the [BluetoothAdapter](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter), you can find remote Bluetooth devices either through device discovery or by querying the list of paired devices.

martinohanlon, 2023-01-02:
Thanks a lot @ukBaz - really helpful.

Interestingly BlueDot gets the list of paired devices through `getBondedDevices`  https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices() - which according to the docs only needs `BLUETOOTH_CONNECT` 🤷

martinohanlon, 2023-01-02:
I have built another APK which requests `BLUETOOTH_SCAN` as well.

https://github.com/martinohanlon/BlueDot/blob/dev/clients/android/app/release/app-release.apk

If anyone could test that would be amazing :)

Boeeerb, 2023-01-02:
Initial tests are no crashing (again I don't have a Pi to test with) but Android side is all good now - So will hand over to someone who has a Pi and software to check connection side of things

Merisiel0, 2023-01-02:
I've tested it with android 12 and the latest raspbian os, it doesn't crash anymore, thank you all!

martinohanlon, 2023-01-03:
Thank you. I have just submitted version 2.2.1 of the Android app to Google Play. It normally takes 24 hours to review before rollout starts.

chadaeschliman, 2023-01-03:
Got 2.2.1 from the play store, fixed the issues for me.  Thanks!

martinohanlon, 2023-01-26:
Closing as I think this was resolved with v2.2.1 of the android app.

closed 2023-01-26 by martinohanlon