martinohanlon/BlueDot #182: can't bluedot be used in devices other than raspberry pi ?

InfictionLabs, 2023-01-02:
Fill in the details or delete as appropriate :)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

or 

Run this program
```python

```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Version [e.g. Buster]

**Additional context**
Add any other context about the problem here.

martinohanlon, 2023-01-02:
Define device?

The bluedot python library will/should run on machines running Linux with a
Bluetooth backend (e.g. blues). e.g. it would run a laptop running Ubuntu.

virtadpt, 2023-01-03:
@martinohanlon Can confirm.

InfictionLabs, 2023-01-05:
we are using imx8m mini som with a yocto build. installed bluedot  using pip3 install bluedot.
installation worked successfully. but when we run the example code it shows the following error

  File "/root/test.py", line 2, in <module>
    bd = BlueDot()
  File "/usr/lib/python3.9/site-packages/bluedot/dot.py", line 844, in __init__
    self.start()
  File "/usr/lib/python3.9/site-packages/bluedot/dot.py", line 1175, in start
    self._server.start()
  File "/usr/lib/python3.9/site-packages/bluedot/btcomm.py", line 351, in start
    self._server_sock = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_STREAM, socket.BTPROTO_RFCOMM)
AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'

please help me know if any supporting libraries or packages are missing.

martinohanlon, 2023-01-05:
>AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH

I am pretty sure this is because the build of Python that is being used hasnt been compiled with the relevant bluetooth "extensions" - I am being vague here because this isnt an area of expertise for me.

ukBaz, 2023-01-05:
@InfictionLabs,

I agree with Martin that this is likely to a lower level dependency that isn't there. Yocto tends to be very basic by default.

You might want to check that BlueZ is installed correctly and up and running:
```
service bluetooth status
```

You can also check that you can scan and connect to a device with the Bluetooth command line tool:
```
bluetoothctl
```

You can check that Bluetooth sockets is in Python with the following couple of lines.
```
import socket
s = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_STREAM, socket.BTPROTO_RFCOMM)
```

Once those lower level things are in place there should be a good chance that BlueDot will work as expected.

InfictionLabs, 2023-01-23:
any packages that we installed or build through yocto  showed "module 'socket' has no attribute 'AF_BLUETOOTH" error.

martinohanlon, 2023-01-23:
>any packages that we installed or build through yocto showed "module 'socket' has no attribute 'AF_BLUETOOTH" error.

Yes, I am pretty sure that is because Python has been built without bluetooth support.

martinohanlon, 2023-01-26:
I was looking at something else and came across this issue on another project including some instructions for building Python with Bluetooth support - it might be helpful for you.

https://github.com/custom-components/ble_monitor/issues/125#issuecomment-713355926

martinohanlon, 2023-01-26:
I am going to close this issue as the root cause is not a fault with bluedot. If I can provide any further advice I am happy to.

closed 2023-01-26 by martinohanlon