martinohanlon/BlueDot #184: Using Blue Dot with Raspberry Pi and HC-05

engineerlab, 2023-03-01:
Hello, is it possible to use the Blue Dot with the Raspberry Pi 2 and an external module like the HC-05?
What I have to change in the python programm to go this way? I read alot the https://bluedot.readthedocs.io/en/latest/gettingstarted.html# but didnt figured out, what to do?

Using the python serial module works fine sending and receaving data from the HC-05, but how can I use it witth the blue dot android app for example or the blue dot library?

Thanks in advance.

martinohanlon, 2023-03-01:
To use a Bluetooth device with bluedot it will have to be recognised by the
OS as a bluetooth device.

I have never used a HC04 Bluetooth module, when it's connected does it
appear as a device when you run bluetootctl?

If it isn't recognised, I think you will be out of luck. A low cost USB
Bluetooth dongle would be an alternative solution.

martinohanlon, 2023-05-09:
Closing due to lack of activity, feel free to reopen if there is more to add.

closed 2023-05-09 by martinohanlon