martinohanlon/BlueDot #186: Raspbery Pi Pico W with Bluetooth

Bernd54Albrecht, 2023-07-03:
BlueDot works fine with Raspberry Pi and Python. I like it. But start time for Raspi is about one minute. Why not take the Pico W with the new support for Bluetooth. Proposal: Support for Micro Python; new APP, if necessary.
Thanks again for BlueDot, from my opinion the best remote control for robot cars.

martinohanlon, 2023-07-06:
Its a great idea and already (notionally) on my list, it is however a pretty significant update and it would require the creation of a micropython version of the comms layer `bluedot.btcomm`, as well as I suspect a lot of other changes. It may end up being an entirely different library.

Bernd54Albrecht, 2023-07-07:
Thanks for the swift reply. I was not the only one asking on the news page https://www.raspberrypi.com/news/new-functionality-bluetooth-for-pico-w/ . May I encourage you to do it. I would like to support, but I think I can't.

Bernd54Albrecht, 2023-07-07:
At least, I can advertise for the APP in Germany. As I did for the existing APP
https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/robot-car-steuern-mit-bluedot .

Bernd54Albrecht, 2023-08-09:
Hello Martin,

(probably) the first Robot Car is controlled by Pico W and Bluetooth BLE.
Attached please find two programs based on the examples on github.
I use a 5-digit-code (from earlier projects with HC-12), where the potis
are scaled down to 0 to 31, with 16 for the middle positions for y
(forward/backward) and x (left/right).
Fifth digit is the joystick button.
For the time being, controller is a second Pico W, but I can connect my
smartphone with the APP LightBlue and punch in the 5-digit-code (not very
handy).
I did not succed to connect with MIT APP Inventor/MIT AI2 Companion, but I
am working on it. (never used MIT App Inventor before).
Looking forward to BlueDot for the Pico W. That would be the best solution.

Greetings
Bernd

rvwilliams88, 2023-09-25:
I really wanted to use BlueDot with my pi peco w and I'm disappointed that it uses a lot of modules or classes not currently available in MicroPython (for example __future__ Event in threading).   I'm not an experienced Python developer so I'm not able to code the necessary items.