martinohanlon/BlueDot #187: Sending feedback to android client

NorthBaandee, 2023-09-04:
**Describe the feature you'd like**
There seems to be no way to send messages from the server to the client, for instance the position of a servo, or any error messages from scripted commands. The only messages i can see at the client end is connection status messages.

**Additional context**
Add any other context or screenshots about the feature request here.

Essenbay, 2024-07-05:
Any solution? I have same question

martinohanlon, 2024-07-05:
It is a question which has been asked before https://github.com/martinohanlon/BlueDot/issues/159

You can change the colour, shape and number of buttons on the app. 

BlueDot is intentionally simple, simple UI and simple approach. Adding text would add complexity to the UI and tbh I like it as it is!