martinohanlon/BlueDot #189: Bluedot android app crashes on first launch

avipars, 2024-08-04:
Appears to be related to the android permissions model. After it crashed, i was able to reopen it and approve the Bluetooth permission (nearby devices) and it is fine

Device - galaxy s23 running android 14 

Is the android app open sourced? 

Is it possible to request adding a dark mode to the app (or can I make a pull request to do so)?

martinohanlon, 2024-08-04:
Its the first report I have had of it crashing on first open. 

There was significant change made to the app for Android 12 (I think) when the permissions model changed.

I dont experience the same issue on my Android 14 device. If you can debug that would be useful.

The android client code is in this repo - https://github.com/martinohanlon/BlueDot/tree/master/clients/android

avipars, 2024-08-07:
Cool! 

As it was downloaded from google play, you may be able to see ANR and crash logs in the dashboard of the app. If I have time, I'll try to get the logcat for you

avipars, 2024-08-07:
FATAL EXCEPTION: main
                                                                                                    Process: com.stuffaboutcode.bluedot, PID: 17259
                                                                                                    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.stuffaboutcode.bluedot/com.stuffaboutcode.bluedot.Devices}: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{261c845 17259:com.stuffaboutcode.bluedot/u0a76} (pid=17259, uid=10076) requires android.permission.BLUETOOTH_CONNECT
                                                                                                    	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:4164)
                                                                                                    	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4322)
                                                                                                    	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:103)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2685)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:230)
                                                                                                    	at android.os.Looper.loop(Looper.java:319)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8919)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:578)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
                                                                                                    Caused by: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{261c845 17259:com.stuffaboutcode.bluedot/u0a76} (pid=17259, uid=10076) requires android.permission.BLUETOOTH_CONNECT
                                                                                                    	at android.os.Parcel.createExceptionOrNull(Parcel.java:3069)
                                                                                                    	at android.os.Parcel.createException(Parcel.java:3053)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:3036)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:2978)
                                                                                                    	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:3042)
                                                                                                    	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1873)
                                                                                                    	at android.app.Activity.startActivityForResult(Activity.java:5760)
                                                                                                    	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:712)
                                                                                                    	at android.app.Activity.startActivityForResult(Activity.java:5718)
                                                                                                    	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:693)
                                                                                                    	at com.stuffaboutcode.bluedot.Devices.onCreate(Devices.java:85)
                                                                                                    	at android.app.Activity.performCreate(Activity.java:8975)
                                                                                                    	at android.app.Activity.performCreate(Activity.java:8944)
                                                                                                    	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1456)
                                                                                                    	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:4146)

avipars, 2024-08-08:
Would you be open to putting the android app in it's own repo and adding to fdroid? If not, would you be fine if I fork the repo to do so?
(And maybe add an optional dark theme)?